derrekr/fastboot3DS #1: Build fail

urherenow, 2017-12-24:
Using ubuntu bash, the build errors out with:

Error opening key file fastboot3DS_priv.key
139948605310616:error:02001002:system library:fopen:No such file or directory:bss_file.c:398:fopen('fastboot3DS_priv.key','r')
139948605310616:error:20074002:BIO routines:FILE_CTRL:system lib:bss_file.c:400:
unable to load key file
0+0 records in
0+0 records out
0 bytes copied, 0.001844 s, 0.0 kB/s
make: 7z: Command not found
Makefile:57: recipe for target 'release' failed
make: *** [release] Error 127

profi200, 2017-12-24:
The key stuff is not so much of a problem. You need p7z installed to make release builds.

urherenow, 2017-12-24:
Windows happily continues (just built it) but Ubuntu bash halts on that error. It has to do with SSL, I think...

profi200, 2017-12-24:
I guess it should be mentioned p7z is required for release builds.

urherenow, 2017-12-24:
Thanks! Good stuff!

closed 2017-12-24 by urherenow

profi200, 2017-12-25:
Done in c58ef56.