derrekr/fastboot3DS #3: Boot slots should be tried in order instead of failing when the first one fails.

Cuphat, 2017-12-25:
Similar to how boot9strap checks sdmc:/boot.firm, but can fall back to nand:/boot.firm if the first one fails. I accidentally moved my boot.firm off of my SD card instead of copying it, and I was surprised to see it didn't fall back to the copy on CTRNAND. This is with the first two slots being configured to autoboot, #1 being sdmc:/boot.firm and slot #2 being nand:/boot.firm of course.

d0k3, 2017-12-25:
They should fail in order, but only for *autoboot* slots. I noticed you shoud have set it correctly, but can you recheck this?

Cuphat, 2017-12-25:
Yeah, as far as I can tell it's all configured properly.

https://www.youtube.com/watch?v=kxCJKtNAfWo

d0k3, 2017-12-25:
Okay, looks like that one was my fault. Can you try this build?
https://f.secretalgorithm.com/GknoG/fastboot3dsv1.0-beta-12733574.7z

Downside: the updater will not accept it, I can't sign these builds. For now, you'll have to install in GM9. Or just chainload it. We may think about a better way to provide test builds.

Cuphat, 2017-12-25:
The test build works fine.

profi200, 2017-12-25:
Fixed in commit 1273357.

closed 2017-12-25 by profi200