derrekr/fastboot3DS #4: Broken error screen when autoboot slot fails on quiet mode.

Cuphat, 2017-12-25:
If the autoboot slot fails to load on quiet mode, this is how the error message displays for me:

[image]

d0k3, 2017-12-25:
Okay, ugh. This has something to do with the screen not initializing properly. We'll look into that.

Cuphat, 2017-12-25:
For a bit more information, the menu looks fine after pressing B or Home. :)

profi200, 2017-12-25:
Screens acting up just like with the splash. Need to look into it later.

profi200, 2017-12-25:
Can you or anyone else with the same problem try this build? https://dl.dropboxusercontent.com/s/fjq1546waja2nog/fastboot3DS.firm

Note: This build is not signed. Please use GodMode9 to install it.

Cuphat, 2017-12-25:
I'm still having the same issue on that build.

profi200, 2017-12-25:
I can reproduce it on my own N3DS XL. I thought i had this fixed but apparently not. Working on debugging it.

profi200, 2017-12-25:
Fixed in 1c6190d.

closed 2017-12-25 by profi200