derrekr/fastboot3DS #5: Add a GUI interface to fastboot3ds

sandhoners123, 2017-12-25:
Currently fastboot3ds only has text like a boot manager but I think it should have also have a GUI just like how SysNAND has one

profi200, 2017-12-25:
We are limited in what we can do without slowing things down. We don't have full GPU support yet so a real GUI is not going to happen until we have.

sandhoners123, 2017-12-25:
Why is it so hard. You already have control of BootROM so why not just inject something into the GPU forcefully

closed 2017-12-25 by sandhoners123

Dimensional, 2018-02-01:
Because the 3DS firmware is what handles the code for the GPU. It's like the BIOS on a PC. When you power it on, the bios does some basic stuff, a POST, figuring out what hardware is on there, then starts the Kernel for Windows and passes the hardware information over to it. Windows then handles the gui stuff, not the bios.

For this to work, a complete reverse engineering of the firmware would be required to figure out the syscalls and assembly code and identifiers necessary to call the advanced features of the GPU directly, instead calling it through the kernel and letting the kernel handle the advanced draws. The standard/basic features of every GPU is loading a bunch of ASCII characters on screen, and it's why so many old PCs had lower resolution until you installed the GPUs drivers to increase resolution from 800x400 to 1024x900, and colors from 256 to 32bit.

profi200, 2018-02-01:
The PICA200 doesn't have any text mode stuff like in the old days of PC GPUs. You have to draw everything manually.
We are lacking some documentation of the low level side of the GPU and a proper implementation for it.