derrekr/fastboot3DS #8: [Feature Request] Option to only store fastboot3DS data on NAND

Lykie, 2017-12-31:
I use multiple SD Cards and I'd like not having to set up the boot settings on first switch.

profi200, 2018-01-01:
Although it supports configs on NAND we decided to prioritise the SD card for safety so you can override the config easily on bricks/problems.

Boot into fastboot without SD card inserted and configure it. When you insert a SD card next time the config will be copied to the SD card from NAND.

Lykie, 2018-01-01:
Okay then. For whatever reason, the first time I inserted my other SD Card, it had me reconfigure it.

closed 2018-01-01 by Lykie

d0k3, 2018-01-01:
You may have gotten that wrong, @Lykie. You need to fully configure fastboot 3DS without the SD card, then poweroff (or continue the boot). Only then you got the config on NAND. And this config will be copied to the SD cards you insert.

Lykie, 2018-01-01:
I configured it without an SD Card initially. My second SD Card is just a bit touchy so it probably didn't detect it properly when I inserted it.