derrekr/fastboot3DS #10: Playing and exiting a DSi game causes a reboot

SuperCodingBear, 2018-01-04:
Basically as the title says. I've tested it with all of my DSi games and every time I exit to the home menu they crash.

CrystaltheGlaceon, 2018-01-04:
What CFW are you using?

profi200, 2018-01-04:
Please give more details. Which CFW? What do you mean with "crash"?

SuperCodingBear, 2018-01-04:
Using Luma v9.0

Exiting any DSiWare game causes an instant reboot of the console, which is a quiet crash I suppose. No panic screens, or crashlogs to dump.

So far it seems all DSiWare may be affected, but games i've had this happen with are

-All game and watch games (eshop releases)
-Mister Driller Drill till you drop
-myDiary
-Adventure in Vegas
-Sleep Clock
-Zelda Four Swords
-Pinball Pulse

All other VC games work fine, it seems to be DSiWare games only that this affects. Sorry for the lack of detail in my initial post, first time bug reporting.

Video here: https://youtu.be/hrGn1ckeJSE

profi200, 2018-01-04:
That's expected behavior. The system is doing a complete hardware reset when exiting DSi or GBA mode. It should also show that splash screen rebooting from GBA mode.

If you don't want to see the splash screen on DSi/GBA game exit enter the fastboot3DS menu holding HOME at boot and set the boot mode to quiet at boot setup --> change boot mode.

closed 2018-01-04 by profi200