derrekr/fastboot3DS #11: Luma3ds does not boot correctly after crash

cdubthecoolcat, 2018-01-06:
After the system crashes, luma3ds boots, loads the splash screen, but then it's just a black screen and never boots to the home menu.  I have tested this with [boot9strap](https://github.com/sciresm/boot9strap) and luma3ds boots up correctly after a crash, which leads me to believe it's related to fastboot3ds.

CrystaltheGlaceon, 2018-01-06:
Can you explain the steps needed to cause the crash?

cdubthecoolcat, 2018-01-06:
The crash isn't the issue, it doesn't matter how you crash it.  The issue is that luma isn't booting up correctly after any crash.

d0k3, 2018-01-06:
Luma's exception handler stores the exception data somewhere in FCRAM (I think), then reboots, and Luma itself detects the exception data and shows it. Maybe @AuroraWright can shed some light?

d0k3, 2018-01-06:
@cdubthecoolcat - more info needed. Do you still see the exception handler screen? What splash screen displays? The Luma one or the fastboot3DS one? And what happens if you set the fb3ds bootmode to quiet?

CrystaltheGlaceon, 2018-01-06:
It would be helpful to at least give an example that we can attempt to replicate.

profi200, 2018-01-06:
With the infos given i can't reproduce this on old 3DS.

cdubthecoolcat, 2018-01-06:
I'm using a new 3ds on luma v9.0 and the current beta release of fastboot3ds on quiet mode.
Steps to reproduce:

1.  Power on console -> Luma splash screen loads -> boots to home menu
2.  Crash the console.  I have tested it by using FBI to launch MINT (titleid 000400300000CE02.  I don't really know what it is, I just know that it crashes the system when launched).  I have also tested it by using a bad ntr cheat.
3.  Press a button to power off the console.
4.  Power on console -> Luma splash screen loads -> black screen indefinitely.

profi200, 2018-01-06:
Reproducable on N3DS XL. Not on o3DS.

profi200, 2018-01-06:
Fixed in 2f277db8.

closed 2018-01-06 by profi200