derrekr/fastboot3DS #14: FastBoot3DS install with a GodMode9 Script

IvanMolina99, 2018-01-31:
Hi!

I have a problem installing FastBoot3DS with a GodMode9 Script.

I know that the best option is install It with OpenFirmInstaller but I need a way to install It directly to the firm1 partition with my GM9 script.

I tried to install It directly with GodMode9 and It works but I tried several times to inyect It directly to the firm1 with my script and It always throw the blue screen.

Anyways, thanks for this amazing proyect, it's a good Boot9strap alternative

profi200, 2018-01-31:
Why firm1: ? This is rather something for @d0k3 and not a fastboot3DS issue.

d0k3, 2018-02-01:
@profi200 is correct - this issue does not belong in here. But, I'll help.

fastboot3DS lacks the sighax signature, so, when you try to set it up via a script, by just copying to firm0 or firm1, you need to inject the signature yourself. The signature is 0x100 bytes @0x100, and GodMode9 (f.e.) includes it. In GM9 scripting you can `inject` data from a file to a file.

closed 2018-02-01 by d0k3