derrekr/fastboot3DS #15: Force restore a Nand.

laramie14, 2018-02-08:
I have tried restoring a corrupt nand which is now not installed. ( My o3ds I gave to a friend came back with no nand on it?) So I have a nand backup just no way to restore it. A dedicated feature to force restore a nand would be nice.

profi200, 2018-02-08:
It does have force restore. You need to enable dev mode for it to work.

laramie14, 2018-02-08:
@profi200 ai tried using the provided force restore but it detects that there is no nand installed and panics. Which I agree is noob proof for sure but it doesn't have a bypass mode. I enabled Dev mode too.

SirNapkin1334, 2018-02-09:
Did he format the NAND via a hardmod? Worst case, you may need to use an eMMC reflasher (basically fixing a Gateway Brick).

laramie14, 2018-02-09:
No, he said he was playing a game, it was close to dying so he turned it off and put it away. When he came back a few days later it was like this? Only thing I can think of is maybe someone (a young one) got to it. I was thinking a rbp reflash with the hardmod

laramie14, 2018-02-12:
I have tried a handful of solutions,non working. When I try to write to the S partition at all it just fails. I have a backup from the console and I pretty much made a d9 script to detect the nand.bin on my SD card, allowed my s part. To be altered and then tried flashing the backup in place of the only thing on my s part. Nand.bin. which fails . I tried booting an emunand to boot around sysnand then fix it but since firms 0/1 arnt in the S part. It doesn't work.

d0k3, 2018-02-12:
Okay... this is new (as in, no known case before yours) and it's bad. It sounds more liek broken hardware than anything else. And, it's not a fastboot3DS issue, it's an issue with your console.

I suggest we still leave that open, trying to get you some help and research this more. Maybe move it elsewhere later. Is there any other info you think could be valuable to us? What do you get when you boot the console? Blue bootrom error or blackscreen?

laramie14, 2018-02-12:
It is definitely a brick, blue screen,bootrom code: 8046, error code : 00F800FE. What I'm thinking is the nand chip isn't populated with anything except a virtual nand.bin and since it's not populated, it can't boot anything. My conclusion is either a hard mod or to find a a way to repopulate it. I have no idea why it's "locked" (what it seems) it does have a hard mod again but that doesn't seem to be working atm.

d0k3, 2018-02-12:
No, that's not it. Your NAND chip is not accessible. The NAND driver returns zero size for the NAND chip, causing the panic message you've seen. It's not available memory populated by something wrong, it's the absence of detectable memory. How big is the `nand.bin` in the GodMode9 `S:` drive? It's zero, right?

Also, the bootrom error misses two lines. Is that all that is displayed?

laramie14, 2018-02-12:
Well actually the nand.bin size is 943mb. Just like it should be on the S part.

laramie14, 2018-02-12:
You can check my Twitter I uploaded some pictures. @MartinezLaramie https://twitter.com/MartinezLaramie/status/962854017440268288?s=09

d0k3, 2018-02-12:
Okay... you also tried copying the NAND backup to `S:`, without the script?

You could still try booting NATIVE_FIRM directly from fastboot3DS or GodMode9, but you need the correct one (for your OS version) and I can't help finding it. I doubt it will change things, but this way you will at least get around a defective FIRM partition if that would be the case.

It's better to wait for a bit and see if something better comes up (instead of stumbling through the dark)

d0k3, 2018-02-12:
@laramie14 - a suggestion. Try IRC. You may not get help immediately, but there are capable people around, and your chances there should be best. Try freenode #Cakey and efnet #3dsdev (yes, both). Just lurk around for a bit, and come back several times during the next few days.

profi200, 2018-02-13:
Keeping it open until solved. But it is clear this is not an issue in fastboot3DS or GodMode9.

d0k3, 2018-03-23:
As @profi200 said, not a fb3ds issue, but rather a strange hardware quirk. @laramie14 - can you tell us how / if you got this solved? Intending to close this issue, but I'd very much prefer to know how this story ended.

laramie14, 2018-03-23:
@d0k3 I haven't had time to open it up, connect the nand chip to my rbp to manually fash the nand to the chip again since the USB adapter card mod is not functioning. I was busy working on my switch 😅 sorry. This can be officially closed. I won't close it with is post since I'm not sure if you'd rather have this open still or not.

d0k3, 2018-03-23:
Alright - just for the record, the issue behind this was in the hardmod. Closing it now...

closed 2018-03-23 by d0k3