derrekr/fastboot3DS #16: Multiple Feature Requests

SirNapkin1334, 2018-02-09:
I had a few feature requests that I posted on the fastboot3DS thread on the temp, but I think they went unnoticed, so I'll just repost them here where they will be noticed.
1. We should be able to cancel NAND backups. I know, this isn't top-priority, but when I tested the NAND backup feature and found that I couldn't cancel it like GM9. Of course, if you cancel it, it'd better delete the file, because a half-NAND backup would be a horrible thing to have on your SD.
2. We should be able to change the NAND backup output path. I don't like the current `SD:/3ds/` output, so I think we should be able to change it, perhaps via the config,
3. We should be able to change the config location. I think the config should be located on the root of the SD, but perhaps it should be changeable on first boot.
Custom splashes without compiling? It'd be nice, perhaps having a file on the SD card like with Luma splashes.
Well, that was it. Keep working on this great program!

SirNapkin1334, 2018-02-09:
Also, fastboot3DS appears to support A9LH payloads. I'll do a bit more research, but when I was selecting a firm to boot, I noticed that firm0 and firm1 backups that I had made some time ago were available to be used. Not sure if this is because it is a firm, just renamed to .bin (since firm0 and firm1 are .bin's), or if it actually work with A9LH payloads. Is there a difference between a .firm and a .bin besides extension?

d0k3, 2018-02-12:
Okay...

1. Double check. You can already cancel NAND backups via the B, HOME or POWER buttons.
2. While not a bad idea per se, the current location is the same location as the location of the config file. It's simple, everyone knows where their NAND backups go, and thanks to the timestamp, everyone also knows the order of their NAND backups. You're also free to copy your NAND backups anywhere you like.
3. Ever heard about the hen-egg-problem? If the path of the config is a config item, where should fastboot3DS pull it from if it doesn't even know the config path before loading the config? No way around this, the config path needs to stay fixed.

Custom splashes are already in the current nightly and will make it to the next release. Also, a FIRM0 / FIRM1 backup is a .firm. No suprise it works

closed 2018-02-12 by d0k3