derrekr/fastboot3DS #17: [Feature Request]  Bonus drive (from gm9) support.

nl255, 2018-03-22:
I would like to request that support for gm9's bonus drive be added to fastboot3ds.

profi200, 2018-03-22:
Unsure if this is a good idea. Not all NANDs leave enough room for it.

SirNapkin1334, 2018-03-22:
Yes, I was going to ask this but forgot to. It's a great place to store things if you don't want to keep it on CTRNAND. Although, some FIRMs don't like to be launched from the Bonus Drive (like Luma3DS).

You could check for the NAND chip size? Perhaps the CID has a relationship to it.
Also, I don't have much proof for it, but I think maybe all n2DSXL systems have a 620MB bonus drive. The first one I owned (accidentally broke the screen but had it replaced under Costco's extended warranty) had a 620MB Bonus Drive, the one I currently have has a 620MB bonus drive, and somebody asked about the function of it because they had a n2DSXL with a 620MB bonus drive. We need some more research on it, though.

nl255, 2018-03-23:
I meant support for using an existing bonus drive created by gm9, not creating a new one.  Presumably gm9 adds a new partition or at least some kind of start/end marker so that it knows how large the bonus drive is.

d0k3, 2018-03-23:
@nl255 - we have our doubts this is a good idea. fastboot3DS does not contain a full fledged file browser, but rather a file selector. The bonus drive is never large enough for a NAND backup, and FIRMs don't take as much space to warrant not just putting them on CTRNAND. Plus, not all systems even have room for a bonus drive.

Can you tell us a good reason why it would make sense to add the bonus drive support?

profi200, 2018-03-24:
And yeah, as mentioned above things like Luma will not work with this. Luma can't and won't handle executable paths except sdmc:/ and nand:/. It also doesn't support twln:/ and twlp:/ (it could though. Very easily but it's not our decision to add that).

d0k3, 2018-03-30:
Okay, there was no activity for a week, so we're closing this. Adding bonus drive support is not totally out of question, but they are just no good enough reasons for it yet. If you can think of one, @nl255 - feel free to contact me.

closed 2018-03-30 by d0k3