derrekr/fastboot3DS #19: [v1.1-beta] Configuration doesn't get copied to SD Card

Lykie, 2018-06-17:
**Steps to Reproduce:**
1. Delete fastbootcfg.txt from SD
2. Turn on 3DS
3. Turn off 3DS (or run any mode that forces the 3DS to reboot)
4. Fastboot3DS menu opens
5. Open a File Manager and see that a blank fastbootcfg.txt has been created in the proper location

**What I think is happening:**
Fastboot3DS boots the 3DS normally
Reads the config from NAND
Makes a config on the SD Card (as if it didn't exist on NAND or SD)
Continues the boot process
Upon rebooting it sees the blank config on SD and enters the menu

Additional info can be provided if required.

profi200, 2018-06-17:
Are you sure there is any config on NAND and it contains settings? Check that using GodMode9.

Lykie, 2018-06-17:
There is a config on NAND containing settings. Manually copying it from NAND to SD works fine.

d0k3, 2018-06-17:
Where on NAND is it, @Lykie, what's the exact path?

Lykie, 2018-06-17:
CTRNAND:/fastboot3DS/fastbootcfg.txt

d0k3, 2018-06-17:
Alright, thanks for teh quick reply. What if you boot your 3DS without an SD card inserted - will it accept the config from the SD?

Lykie, 2018-06-17:
From the NAND you mean? It accepts the config when the SD Card isn't in. It also loads it while it's in and there's no config on the SD.

d0k3, 2018-06-17:
Alright. Yeah, something is wrong here. One more... is this with 1.0 stable or 1.1 beta? Or some other version?

Lykie, 2018-06-17:
1.1 beta (as stated in the title)

d0k3, 2018-06-17:
Ah, sorry. Missed that.

Lykie, 2018-06-17:
Strange. I don't think there's SD Card corruption since I formatted it about 3 hours before this issue happened.

I can take a video of it happening if it helps.

Config
`BOOT_MODE = Quiet
DEV_MODE = Enabled
BOOT_OPTION1 = nand:/rw/luma/release.firm
BOOT_OPTION6 = firm1:
BOOT_OPTION6_BUTTONS = X
`

SirNapkin1334, 2018-06-17:
@Lykie Yes, getting a write permission prompt when deleting the Nintendo 3DS folder is normal - it has savedata and is thus protected under level one of the write permission System (check GM9 README for more info on what is protected). It’s just a safety check to insure you don’t accidentally delete it (as I have done once....).

Lykie, 2018-06-17:
Okay, makes sense.

Anyway, the same issue occurs on all three SD Cards.

The Wii and Wii U SD Cards were formatted by GParted, and the 3DS SD Card was formatted by GodMode9.

d0k3, 2018-06-17:
Hold in, I have an idea... did you by any chance reinsert the SD card while you were navigating the fb3ds menu?

Lykie, 2018-06-17:
I inserted the SD Card while the 3DS was powered off in most instances.

The only times I inserted the SD Card while it was powered on, was while I was deleting the config file through GodMode9.

d0k3, 2018-06-17:
Alright, I think I've solved it. Can you try this testbuild?
https://f.secretalgorithm.com/DzqTm/fastboot3ds.firm

Only installable via GM9, but I guess you know.

Lykie, 2018-06-17:
That fixed it!

closed 2018-06-17 by Lykie