derrekr/fastboot3DS #24: cant get my console to boot after switching sdcards and I know its something with fastboot

RyanmFisher, 2018-07-23:
after swapping sd cards to a larger one fastboot keeps giving me this error saying that a directory that one of my boot slots is set to does not exist though when looking at my fastboot settings there is no boot slot setup there and its basically just telling me that it cant find a firm where I never told it to look I tried just launching luma from there and I just get a black screen after launching deleting the settings file from the sd card did not help

d0k3, 2018-07-23:
Tbh, you should have used proper punctuation. This is a little hard to read. Can you make some photos and upload here to show us what happens?

Also, two things are almost sure... (1) your fb3ds settings are configured to look for some firm where none exists and (2) you most probably didn't launch a compatible version of Luma3DS (at least in case it worked earlier).

RyanmFisher, 2018-07-23:
heres an image of my settings https://cdn.discordapp.com/attachments/265325347824730114/471048661845934100/image.jpg just replaced the boot.firm with a freshly downloaded one still getting an error on boot

d0k3, 2018-07-23:
That looks good, but you should format the SD card again, I guess (use GM9), and also redownload Luma.

RyanmFisher, 2018-07-23:
cant access godmode9 holding start on boot just gets me a black screen and I have the payloads on my sd

RyanmFisher, 2018-07-23:
strange I just decided to reinstall using ntrboothax and that seemed to fix the issue

closed 2018-07-23 by RyanmFisher