derrekr/fastboot3DS #25: 11.8 support?

smurfhunter, 2018-08-06:
Does this work with 11.8? I seem to be unable to boot at all after updating to 11.8. Complete black screen. No splashes.

CrystaltheGlaceon, 2018-08-06:
OFW updates won't effect FB. Did you make sure you updated your CFW after updating to 11.8?

smurfhunter, 2018-08-06:
Yeah, I updated my CFW, but I get a solid black screen now.

smurfhunter, 2018-08-06:
I have the latest Luma and Rei-six and no button combos get me into either of those or the settings for fastboot3ds

smurfhunter, 2018-08-06:
Any suggestions on what to do here at all?

CrystaltheGlaceon, 2018-08-06:
I don't really have an answer, unless the FIRM0 protection broken for you CFW. Which is unlikely as that would have just removed FB3DS all together.

RShadowhand, 2018-08-06:
Every CFW out there protects FIRM0/1. He might be having a screeninit issue like I did in closed beta.

smurfhunter, 2018-08-06:
Anyway to tell if that's the case?

RShadowhand, 2018-08-06:
Have a fastboot config that autoboots immediately to whatever your CFW is. If it boots, you're having a screeninit issue.

d0k3, 2018-08-06:
@smurfhunter - keep in mind OFI does not have screeninit, so you better put boot9strap or GodMode9 on the ntrboot cart.

Now, it's basically impossible fb3ds is at fault for this. Maybe something got wrong when you updated to v11.8? You can also check if Luma still works by holding SELECT on boot. Also try what @RShadowhand said.

profi200, 2018-08-06:
I don't think this has anything to do with fb3DS. A simple update won't brick/remove it unless you ran vanilla NATIVE_FIRM at update time (not the case with any CFW).

We need more details.

smurfhunter, 2018-08-06:
@RShadowhand where is the config for fastboot3ds? I had it set to do quickboot so it should be displaying the FB3DS splash if it works and screeninit works, so there's got to be an issue there.

I use reisix with luma as a backup, it should be autobooting reisix. When I ran the system update, I was in reisix, and upon the system rebooting due to a complete update, I got this black screen. The only output I get from the system is a blue power light. None of my other button combos work and it does not appear to autoboot reisix or luma (I renamed luma's firm file to the autoboot's name- nothing).

Thanks for the help, it's appreciated. I should be able to flash my flash cart later today with GM9 or something else.

smurfhunter, 2018-08-07:
Thanks for the help guys, I fixed it by reinstalling to firm0. My assumption is rei six didn't protect firm 0. Feel free to close this :)

d0k3, 2018-08-07:
Maybe you want to open an issue for Rei Six then.

closed 2018-08-07 by d0k3