derrekr/fastboot3DS #28: [Enhancement] Wider display of path and filename in boot menu

jelbo, 2018-11-20:
See title. I have multiple .firm's to choose from, but the names are not fully displayed because the display width is limited. There is enough space on the screen though. Please widen this display.

d0k3, 2018-11-25:
Can you provide us with a screenshot? Photo is actually enough.

jelbo, 2018-11-26:
Boot slot selection shows a truncated path and filename view:

[image]

Same for the initial boot slot setup screen:

[image]

Boot from file shows a wide file list. Maybe this can be used as a basis?

derrekr, 2018-11-26:
Thanks for the photos!
We found the code that's causing this issue and we are going fix it.

d0k3, 2018-11-26:
Alright, I increased the limit. If you want to try it, here's a test build:
https://f.secretalgorithm.com/Vas9O/fastboot3ds.firm

commit 92579d68 (2018-11-26): Increase size limit for path in boot setup.

Fixes #28
closed 2018-11-26 by d0k3