derrekr/fastboot3DS #31: [Question] (auto) nandwrite disable? config writing to nand

codeM-X, 2018-12-12:
I am currently trying to modify FB3DS. I want to disable FB3DS to write config files to NAND even if installed to firm0, but if there is a file (self created) it should be able to boot from it. I should not lose any functionality.
So can i just delete [source/arm11/config.c#L172](https://github.com/derrekr/fastboot3DS/blob/92579d68f2fcb5780622a7d2ab6613c4a5771c0b/source/arm11/config.c#L172) to [source/arm11/config.c#L180](https://github.com/derrekr/fastboot3DS/blob/92579d68f2fcb5780622a7d2ab6613c4a5771c0b/source/arm11/config.c#L180)
without breaking anything? Is this the only nand autowrite?

codeM-X, 2018-12-12:
Okay I tested it, now fasboot3ds wont create a file in NAND:/fastboot3DS/ but it also wont load a config file store in it. How can i change that?

d0k3, 2018-12-12:
In general, how to handle configs from CTRNAND / SD has already been discussed in depth, and each  approach has it's up- and downsides. Unless such a late change would benefit (almost) all users, we do not intend to change it for everyone.

I may have a look at your changes later and see about how you can achieve what you want to do, but it most likely won't make it to the master branch.

codeM-X, 2018-12-12:
Yeah I accept that! I just want to change it for my fork of fastboot3ds.

profi200, 2018-12-12:
Is this for a personal build (only you will use it) or a full blown fork? If the later it would be good if you adjust the name of your fork so it doesn't conflict with this one. And you need to generate your own key pair for signing these (see arm9/data and signFirm.sh).

codeM-X, 2018-12-12:
its just a personal build. I just wanted to remove the auto config creation. I am a little a freak in this behavior. I think no homebrew should write to the nand if not forced by the user. I think I have got the cleanest solution now, without breaking any future of FB3DS, even config loading form nand works, it just needed to be created manually with GodMode9 in nand:/Fastboot3ds/.

codeM-X, 2018-12-13:
OT: Fist, I have re-forked FastBoot3DS to prevent GitHub to push pull requests to the “main branch”, so you wont get spammed by the changes I made anymore, sry for that inconvenience that was not intended (I missed click). Furthermore I compile with Linux so I don't have to remove the -”flto” flag from the “make-file”.

@d0k3 
The change I actually made is quit simple. It worked I have tested it.

I changed [!main branch! /source/arm11/config.c#L178](https://github.com/derrekr/fastboot3DS/blob/92579d68f2fcb5780622a7d2ab.6613c4a5771c0b/source/arm11/config.c#L178) to [!fork! /source/arm11/config.c#L178](https://github.com/codeM-X/fastboot3DS/blob/5238f08d65e97c586e3a647e936a3c5194bfc7d3/source/arm11/config.c#L178)

You could also replace the original line (main branch) `createFile = true;` with `adpotChanges = true;`, but I think setting the variable is less work for the cpu, so I made it simpler with `filepath = NandFilepath;`

Explanation: 
By removing `createFile = true;` no file is created in the NAND even if no SD-card is inserted. But for some reason removing the whole if condition [main branch /source/arm11/config.c#L176](https://github.com/derrekr/fastboot3DS/blob/92579d68f2fcb5780622a7d2ab6613c4a5771c0b/source/arm11/config.c#L176) won't change anything so I replaced  !(main branch) `createFile = true;` with `adpotChanges = true;` ! OR !(main branch)  `createFile = true;` with `filepath = NandFilepath;` !. I prefer the latter one to prevent errors at a letter time coused by `adpotChanges = true;` (is something gets changed about `adpotChanges = true;` .

What I tested on real hardware after compiling and injection the .firm:
Config loading from NAND works with fastboot3DS.
Config editing from NAND work with fastboot3DS.
(AUTO-)Config copying from NAND to SD with fastboot3DS works.
No config is created by fastboot3DS on the NAND if no SD is inserted.
No changes will be made by fastboot3DS on the NAND if no SD is inserted, except there is already a file on the NAND.

d0k3, 2018-12-13:
"For some reason" is not a good approach when programming :^). Try a clean solution.

Replace this whole thing:
https://github.com/codeM-X/fastboot3DS/blob/5238f08d65e97c586e3a647e936a3c5194bfc7d3/source/arm11/config.c#L155-L180

With just this:
https://github.com/codeM-X/fastboot3DS/blob/5238f08d65e97c586e3a647e936a3c5194bfc7d3/source/arm11/config.c#L157-L170

Also, granted, I don't liek it much, because there is no proper handling for when there is no SD card. It's pretty okay for your personal use, though.

Can we consider this solved then? If so please close it.

closed 2018-12-13 by codeM-X