derrekr/fastboot3DS #34: compatibility issues with Luma3DS

Ty-Dye, 2019-03-22:
The latest commit (474eb30) on Luma3DS dose not boot on fastboot3DS but dose on B9S and GM9 (according to @AuroraWright).

I have a N3DS on system 11.9 as well as fastboot3ds 1.2 stable

Thanks for reading, I just wanted to inform the developers about compatibility issues

CrystaltheGlaceon, 2019-03-23:
This was actually discovered by me this morning, but I was bit tired and didn’t feel like reporting it until I got some sleep. Moving on.
So it appears the latest commit doesn’t boot with fastboot. I tested this on both my old and new 3DS, both result in a softlock of just a black screen and blue LED. I can not replicate this issue when booting the build on B9S and GM9, tested again on an old and New 3DS

Ty-Dye, 2019-03-23:
@CrystaltheGlaceon yea, i just upgraded to fastboot today and I was planning todo so with luma as well (i was and am now on 9.1 official)

how could the devs (of fastboot or luma) fix this

profi200, 2019-03-23:
See https://github.com/AuroraWright/Luma3DS/issues/1232 for updates. I think this is a Luma bug.

CrystaltheGlaceon, 2019-03-23:
I think it might be both actually because Luma3DS still works just fine with GM9 and B9S, it's only on Fastboot that it has an issue. Something tells me that the lastest Luma commit does break the compatibility and it's up to either them or Fastboot to fix it.

profi200, 2019-03-23:
It appears the stack pointer is in a rather bad position which triggers the bug while it doesn't for other chainloaders. There are calls to functions messing with the stack before they are setting the stack pointer up correctly.

CrystaltheGlaceon, 2019-03-23:
Interesting, so which party should fix this?

TuxSH, 2019-03-23:
Fixed on my end, please check if there are other issues.

Ty-Dye, 2019-03-23:
@TuxSH Thank you very much! you do amazing work!

CrystaltheGlaceon, 2019-03-23:
@TuxSH
I will do some testing later tonight

Ty-Dye, 2019-03-23:
also im not sure if I should close this, ill leave it open for now just in case

CrystaltheGlaceon, 2019-03-24:
I can confirm that it now officially working again

Ty-Dye, 2019-03-24:
@CrystaltheGlaceon me to

closed 2019-03-24 by Ty-Dye