derrekr/fastboot3DS #35: Compilation issue

Multimegamander, 2019-03-26:
source/arm9/firmwriter.c 
166:28 error: expected expression before '<<' token
  if(vers < ((u32)VERS_MAJOR<<16 | VERS_MINOR)) return UPDATE_ERR_DOWNGRADE;
                            ^~
make[2]: *** [/opt/devkitpro/devkitARM/base_rules:84: firmwriter.o] Error 1

profi200, 2019-03-26:
Do you have git installed inside the devkitPro provided msys2 (pacman -Syu git)? (if you are using Linux install git through your package manager).

Multimegamander, 2019-03-26:
I do have Git

Multimegamander, 2019-03-26:
but, does it need to be part of a git, bc I have it local?

profi200, 2019-03-26:
Did you git clone or did you download a zip file from the releases section? Because the zip files are missing the git files so it can't get the version infos and it will fail on compilation like that.

Multimegamander, 2019-03-26:
oh, I downloaded

Multimegamander, 2019-03-26:
it worked, my bad :P

closed 2019-03-26 by Multimegamander