derrekr/fastboot3DS #37: Default dump path shouldn't be sdmc:/3DS/

ParzivalWolfram, 2019-08-23:
sdmc:/3DS/ is where things like HBL payloads and homebrew app data goes, it would clutter up the folder further. I would propose a system as such:
NAND backups: sdmc:/backups/
boot*/otp dumps: sdmc:/dumps/

SirNapkin1334, 2019-09-02:
Well, there's already an sdmc:/fb3ds folder, isn't there? Maybe those can be a subfolder of that folder.

derrekr, 2019-09-08:
We think that the majority of our users prefer the current path. Build the project from source if you would like to use a different path.

closed 2019-09-08 by derrekr

ParzivalWolfram, 2019-09-08:
Silence isn't always agreement, but okay.

profi200, 2019-09-10:
This probably got lost. We actually discussed this a while ago and came to the conclusion, that a path change may break for example tutorials or similar and is therefore not an option. I also think another folder at the root is not a good thing.
Are there even enough homebrews around that throwing files in /3ds is a problem? The only realistic approach is a folder inside /3ds imho.

ParzivalWolfram, 2019-09-10:
That would work, but for those using lots of 3DSX homebrew, just dumping
shit to /3ds would further muddy up things and potentially stop allowing
GM9 to manage everything in there.