derrekr/fastboot3DS #39: Boot from bonus drive support

ParzivalWolfram, 2019-11-01:
I'd like to put a few utilities on my NAND's bonus drive and boot from there, however I can't seem to find it in 1.2 stable despite it being formatted, visible and read/write in GM9. Would there be a way to add this support or am I just missing it?

d0k3, 2019-11-01:
Wouldn't be too difficult to support this in theory. Up to @profi200 and @derrekr if we want to do it.

profi200, 2019-11-02:
It would not be very hard to implement but:

Reasons against this:
* Increases boot time (need to mount another drive).
* It's not compatible with for example Luma3DS (it can't mount and access its config from the bonus drive (argc/v)).
* There is basically no demand for such a feature (this and #17 are the only requests so far).

Therefore we decided to reject this.

closed 2019-11-02 by profi200

ParzivalWolfram, 2019-11-03:
For the boot time issue, maybe add a line to the config to enable/disable
this feature?
As for the rest... fair points, I suppose.

profi200, 2019-11-03:
Then we have to add another setting and the problem of users complaining in case they want to boot Luma from there (which will not work correctly). But to be honest Luma will not work from twln/p either but we support it.