derrekr/fastboot3DS #40: Unable to load 100mb+ files

PlantBeatdown, 2019-11-20:
I use a forwarder to put NDS games onto my 3ds. Fastboot3ds messes with the loading of larger games though, as it'll flash the Fastboot3ds logo and take me to the screen where you turn off the system. I have found no workaround, and it'd really suck if I'm unable to continue with my playthroughs because of this;;

d0k3, 2019-11-20:
Are you using the most recent version? We really don't touch FCRAM in fastboot3DS, so I'd be surprised if that was the cause.

PlantBeatdown, 2019-11-21:
As far as I know, I am using the latest version.

d0k3, 2019-11-21:
Can you give us a more detailed description then? What are you trying to do, what happens?

What memory areas are used and where do you think fb3ds writes?

profi200, 2019-11-21:
We need much more detailed infos.

* Which 2/3DS model?
* What are the fb3DS settings (you can paste the /3ds/fastbootcfg.txt file)?
* Which version of Luma3DS (or what else you use if not Luma)?
* How did you create these forwarders and what do they forward to? nds-bootstrap?
* Tried to reformat your SD card (after making a backup)?

It being ROM size dependent makes no sense to me since this seems to happen before TWL_FIRM even runs.

profi200, 2019-12-03:
@PlantBeatdown 
Reminder. If we don't get an answer we will close this.

profi200, 2019-12-09:
Closing this because no answers.

closed 2019-12-09 by profi200