derrekr/fastboot3DS #41: Config saving behavior

Z7QsbkpL5T, 2019-11-27:
Right now the behavior seems to be:
- If SD is present, load and save there
    - If there's no existing config, look for one on the NAND
        - **Save to SD even if loaded from NAND**
- If SD is not present, load and save on the NAND instead

That way I cannot force it to use the NAND even if an SD card is present.

Can the behavior be changed so that it saves to the NAND if the config was loaded from the NAND?

FlameKat53, 2020-01-04:
kinda related to this issue, i guess. 
i think we should save the config to nand on ctrnand:/rw/fastboot3DS/ and not to ctrnand:/fastboot3DS/ of the NAND

profi200, 2021-04-08:
It's designed this way to prevent bricks. The eMMC is not something you can take out and stick in your PC. If something goes wrong on there you have to get a ntrboot flashcard or hardmod to unbrick that. There is also no way we can save the config location because we are relocating the config where this would be saved in.

It was/is a difficult decision.

NintendoManiac64, 2021-04-08:
OK, brink prevention is a very legitimate reason for the current behavior.

I think my confusion would have not occur if it was somehow made more apparent to the user that there are effectively two modes with regards to saving.

In particular, I thought settings were always saved to the SD card regardless, and therefore if there's no SD card then settings can't be saved at all and it just always loaded the default (this is akin to how games on consoles with memory cards work, like the GameCube).

d0k3, 2021-04-09:
Actually, the config saving behavior is changed in a recent nightly, allowing you to have your config on NAND only, while still keeping the possibility to override it with a SD config. Maybe try a recent nightly?