derrekr/fastboot3DS #42: Splash occasionally appears messed up.

FlameKat53, 2020-02-06:
This happens at least on commit c417a8a4752779879af78253da21171c09e1ae92, though i do not know if it happens in ef732dc8da6c3ab4ffffe0e1629feeceeb347796, the latest commit at the moment.

redunka-zver, 2020-02-07:
Can confirm that it happens with the latest commit as well.
The splash appears randomly corrupted when fb3ds boots from FIRM0, but it always seems to look normal if fastboot3DS.firm is chainloaded through something else first (like GM9 or Luma).
O3DS here, if that matters.

profi200, 2020-02-07:
Not able to reproduce this. I need more infos.

When does it corrupt? While entering the menu (through the HOME button) or quick boot?
Does it happen with custom splash (on the SD) or with builtin aswell?
Did you rebuild it with a different splash baked in?

redunka-zver, 2020-02-07:
I usually use quiet mode, so I originally noticed it while entering menu, but it definitely happens with other two boot modes too.

For quick mode even the "Press HOME to enter menu" text that appears on bottom screen also gets corrupted.

All of that happens with default builtin splash, I've never actually tried using custom splashes (neither from SD nor by replacing the builtin one), but I can try when I have some time later, if that can help.

P.S. Changing splash duration, whether it's shorter or longer, doesn't seem to help either.

profi200, 2020-02-07:
Yeah, i can't reproduce this. It's working on my o3DS with debug and release builds. You are building under Windows, correct? Possibly the link order affects this and changes timing. I built everything under Linux. Can you upload one of these broken builds? I have an idea what the problem could be.

The last commit i just pushed is unrelated btw and can be ignored.

redunka-zver, 2020-02-07:
Yeah, I build under Windows.
Here's that one that I currently have installed:
http://www.mediafire.com/file/tchiu6qs6wabpbq/fastboot3DS.firm

profi200, 2020-02-07:
Same with your build. Ok, a blind attempt at this. Try the attached build.

[splash_test.zip](https://github.com/derrekr/fastboot3DS/files/4170825/splash_test.zip)

redunka-zver, 2020-02-07:
With this test build there seems to be no corruption with both normal mode and while entering menu with quiet mode.

Quick mode still has some minor corruption on both screens…

Does it matter what was used to flash the firm? I used GM9 for both builds, latest commit from the staging branch.

profi200, 2020-02-07:
Ok. All i changed was the 3 ms wait in GFX_init() waiting for VBlank instead. At least this confirms it's timing related.

Doesn't matter.

[splash_test2.zip](https://github.com/derrekr/fastboot3DS/files/4170976/splash_test2.zip)

redunka-zver, 2020-02-07:
No change unfortunately, the quick mode is still corrupted.
But I noticed something weird, with both test builds… When I boot with no SD card the corrupted splash looks different almost every other time, but when the card is inserted it always (at least most of time) looks the same, and the corruption is less severe.
The config is the same on both SD and CTRNAND.

profi200, 2020-02-07:
What about this one? I'm running out of ideas to be honest.

[splash_test3.zip](https://github.com/derrekr/fastboot3DS/files/4171129/splash_test3.zip)

redunka-zver, 2020-02-07:
Same. :(

Actually, the corruption is random almost every boot now, no matter whether SD is present or not.
Though, maybe even last time I was just "lucky" to get the same looking corruption several times, because when I tried to increase splash duration to take some photos it started getting random ones more often.
Oh well, I guess at least I can try to show how it looks like (the third pic is what was appearing the most on last two builds with SD card inserted):

[image]
[image]
[image]

 I'm sorry for not being of much help here, and thank you for your patience.

profi200, 2020-02-07:
That kinda looks like the corruption i was getting a long time ago on N3DS XL which seemed texture copy related and got "fixed" with a workaround. I recently pushed a fix which was supposed solve this properly. Looks like this is an entirely different bug after all and was never properly fixed.

I'm going to take another look into the gsp module because i remember some unknown reg writes which could be related. If that doesn't work i'm going to try another workaround.

redunka-zver, 2020-02-07:
…I'm an idiot.
It completely flew over my head that neither OP nor I have ever mentioned that this bug started to occur exactly after [this](https://github.com/derrekr/fastboot3DS/commit/04efa8e3fd04fef2455ee130c55b0986e18da1dc) commit.
I don't think I've ever seen it happening before that, and I just tried compiling the [commit](https://github.com/derrekr/fastboot3DS/commit/c5884f784a5117fbd429452a3a739b10bc7e1633) before that, and it works just as it should!

profi200, 2020-02-07:
Oh, yeah. [That](https://github.com/derrekr/fastboot3DS/commit/04efa8e3fd04fef2455ee130c55b0986e18da1dc#diff-2624bf69d029c632ce33659badc324b1L295) was the workaround i was talking about. I thought this was the same bug but it's not. I still want to try fixing this correctly.

redunka-zver, 2020-02-07:
Sorry for wasting your time like that. xP
I tested the latest commit you just pushed, and now everything looks perfect!
[image]

profi200, 2020-02-07:
Ok, what about this one? If you have more than one 3DS test it on all of them.

[yet_another_test.zip](https://github.com/derrekr/fastboot3DS/files/4172180/yet_another_test.zip)

redunka-zver, 2020-02-07:
Sorry for late reply, I went to sleep.
That build seems to act the same as old buggy ones: all three boot modes are borked this time.
Unfortunately, I only have one console, perhaps @FlameKat53 can help with more tests?

profi200, 2020-02-07:
I basically only wanted to know if this build works and if it does, does it work on all models? Since it doesn't work the other tests are superfluous. I thought maybe these extra reg writes will do it but unfortunately not :/

closed 2023-11-07 by FlameKat53