derrekr/fastboot3DS #44: [BUG] FastBoot3ds says it dumps the bootroms but doesn't actually dump the bootroms.

TheMasterOfMike, 2020-07-22:
Version of FastBoot3ds: Release 1.2

Issue: Attempting to dump Boot9.bin and Boot11.bin using FastBoot3ds's built in setting says it succeeds, yet doesn't actually dump the bootrom. (Proven by searching through every single directory on the SD Card after dumping)

System Tested on (not including the two systems in NH Discord):

New Nintendo 3ds - FB3ds 1.2+Luma 10.2 and SysNand 11.13.0-45U.

profi200, 2020-07-22:
Can't reproduce this. Neither with version 1.2 nor with a build from master. "boot9.bin" and "boot11.bin" are dumped to sdmc:/3ds (the "3ds" dir in the root of the SD card). Did you make sure the SD card is not corrupted?

TheMasterOfMike, 2020-07-24:
The SD Card I have doesn't have issues (tested it twice), though that being said trying it again did actually dump the bootroms.

Still though, that doesn't explain both why it's not consistent (for me anyways) and why when opening GodMode9 I can't dump the bootroms from there.

(Also just saying dumping to sdmc://3ds/ seems like a bad location to dump it to but I digress)

profi200, 2020-07-24:
Not consistent? It failed once so far for unknown reason. And GodMode9 can't dump it, because the bootroms are not placed in memory like b9s does. It only runs with unlocked bootroms when requested via the menu.

It's /3ds because we didn't want to fill the root with any more files or dirs.

derrekr, 2020-08-08:
Closing this issue now, since we couldn't reproduce it.

closed 2020-08-08 by derrekr