derrekr/fastboot3DS #46: New(er) Release needed

urherenow, 2020-09-29:
It has been a year and a half since the last release. A lot of those 55 commits look really important, and people can't simply build the current tree and install with OFI, without adding the key. Please post a new build and label it as beta, or provide the key info in a text file. I have to ask myself what the point is, of using a key (and requirement for that key) for two entirely open-sourced projects. It's just an annoyance.

profi200, 2020-09-29:
The private key won't be made public as that would defeat its point. We sign the releases, because we want to stop people installing potentially malicious builds through fb3DS itself. Of course we can't stop you overwriting it using GodMode9 for example.

A release is still planned. People have a real life and maybe other interests as well and so we kept pushing this back. Hopefully we will get the release out soon.

urherenow, 2020-09-29:
Thank you! 
I'm on a deployment and haven't been home since May. The way this is written, I can't even compile it myself, because I can't clone it with git (no internet access on my personal laptop). Seems some variables require info that's tucked away in the .git folder that you simply don't get by downloading a zip of the source.

profi200, 2020-09-29:
It's getting the version number from git, yes. For some odd reason Github never includes the git history in the zip files.

urherenow, 2020-10-02:
So, I just reminded myself of why I never tried installing a build. I got around the errors by doing:

make NO_DEBUG=1 VERS_MAJOR=1 VERS_MINOR=2

but the result is a 138.8kb .firm file! Isn't this too big to install in firm0? b9s is 30.7kb

urherenow, 2020-10-02:
I guess it's not too big. I'm also hard-modded AND have an ntr flash card, so I installed it (and noticed that the firm0/1 backups are 4MB). Didn't seem to cause a brick. Using boot_Ctr9 to chainload my GW emunand seemed broken with it, either with a direct chainload or with Luma3ds chainloading it. 

Godmode9 seems to recommend this, but I'm unsure of the advantages.

closed 2020-10-02 by urherenow

profi200, 2020-10-02:
I don't know what these old bootloaders rely on. Possibly they require screens being inited (they must indicate so with a flag in the FIRM file).