derrekr/fastboot3DS #47: Keycombos doesn't work

pokemaniac05, 2020-10-04:
I have a Retail Old2DS with Fastboot3DS v1.2, Luma3DS v10.2 and Nintendo System Software 11.13.0-45E.
I setted 12 boot slots: 6 for NAND and 6 for SDMC. I have following trouble from September 12th (a day after I installed Fastboot3DS) : GodMode9 (booted with [START] key) sometimes starts and other times Luma3DS Chainloader Menu is started. Other payloads, such as Decrypt9 (booted with [A+B+X+Y] keycombo) doesn't anymore.
SDMC config file:

___
BOOT_MODE = Quick
BOOT_OPTION1 = sdmc:/firm_linux_loader.firm
BOOT_OPTION2 = sdmc:/luma/payloads/godmode9.firm
BOOT_OPTION2_BUTTONS = START
BOOT_OPTION3 = sdmc:/luma/payloads/Decrypt9WIP.firm
BOOT_OPTION3_BUTTONS = A + B + X + Y
BOOT_OPTION4 = sdmc:/luma/payloads/ntrboot_flasher.firm
BOOT_OPTION4_BUTTONS = UP + R + L
BOOT_OPTION5 = sdmc:/boot.firm
BOOT_OPTION6 = sdmc:/luma/payloads/OpenFirmInstaller.firm
BOOT_OPTION6_BUTTONS = START + UP + R + L
BOOT_OPTION1_BUTTONS = SELECT + START

___
NAND Config file:

___
BOOT_MODE = Quick
BOOT_OPTION1 = sdmc:/boot.firm
BOOT_OPTION2 = nand:/rw/luma/payloads/GodMode9.firm
BOOT_OPTION2_BUTTONS = START
BOOT_OPTION3 = nand:/rw/luma/payloads/Decrypt9WIP.firm
BOOT_OPTION3_BUTTONS = A + B + X + Y
BOOT_OPTION4 = nand:/rw/luma/payloads/ntrboot_flasher.firm
BOOT_OPTION4_BUTTONS = UP + R + L
BOOT_OPTION5 = nand:/boot.firm
BOOT_OPTION6 = sdmc:/luma/payloads/SafeB9SInstaller.firm
BOOT_OPTION6_BUTTONS = A + B + R + L + X + Y

___

This files are completely created by Fastboot3DS (I didn't edit them manually).
I tried to reinstall Fastboot3DS using its update utility, GodMode9 FIRM0 installer and OpenFirmInstaller(I used it to install Fastboot3DS for the fist time), but I didn't resolve this trouble.

derrekr, 2020-10-05:
Hi pokemaniac05,
Could you try pressing the buttons for your keycombos before powering up your console? Also hold the buttons, until the splash screen disappears.
Basically, if you don't press them as soon as possible, fastboot3DS could miss them.

pokemaniac05, 2020-10-05:
Thanks derrek, now I can enter keycombos and boot all payloads.
But before September 12th this keycombos worked if entered after powering on the console.
May bootmode changing (from quiet to quick and from 1000 ms to 1500 ms) be the cause?

closed 2020-10-05 by pokemaniac05