derrekr/fastboot3DS #51: where is fastboot3DS.firm?

arielfabulous, 2022-07-13:
I am VERY AWARE that this is probably the dumbest issue or question anyone will ever post here, but I just don't care at this point. I just want to get the ROM from Ocarina of Time 3D onto my computer so I can play it on Citra. I am not a computer person. I only know how to follow instructions. 
I have downloaded the .zip file for fastboots3DS provided. The first instruction in the README is the following:
"Have `fastboot3DS.firm` inside the `sdmc:/ofi`folder on your SD card."
I have no idea where this "fastboots3DS.firm" file is. I have searched the folder for this, and can find no such file. Is this referring to the whole extracted folder? I am incredibly confused and do not want to risk bricking my 2DS by doing this trial and error. 
Thanks.

james-d-elliott, 2022-07-14:
There are 3 files in the released archive, it's the top one:

[image]

derrekr, 2022-07-14:
Yes, the fastboot3DS.firm file should be inside the .zip archive you downloaded (just extract it).
Note that fastboot3DS itself cannot dump ROMs from gamecards. However, you can use GodMode9: https://github.com/d0k3/GodMode9/

amplesand, 2022-09-04:
@arielfabulous I recommend that you simply go to <https://3ds.hacks.guide/>, set everything up, and then look through [this](https://3ds.hacks.guide/dumping-titles-and-game-cartridges) on how to dump your game using GodMode9 (which should be installed after following the CFW guide).

closed 2022-09-04 by derrekr