derrekr/fastboot3DS #52: how to boot 3dsx file with key combo or slot config

Ixolaro, 2022-07-17:
can you add support to boot directly dsx files from boot slot, line unlaunch on dsi

amplesand, 2022-09-04:
@Ixolaro That isn't really something in the scope of this program. Though theoretically possible to do, from my understanding it would be a pain and hasn't been done yet because increasing the ease of opening HBL is just not worth the work for that extra 30 seconds of unattended device loading. Just use the HBL like everyone else.

MiksFish, 2025-04-11:
just set luma to autobot hbl

Anonymous941, 2025-04-12:
You can't really directly boot 3dsx files in the same way as NDS files because the 3DS has a complicated OS that takes over and runs through various boot stages before launching any app.

Luma3DS has an autooot homebrew launcher option, and you can put whatever app you want as `/boot.3dsx` (or change the configuration file), but it doesn't let you use a key combo.  You could make a feature request for Luma3DS to add that if you want

MiksFish, 2025-04-28:
you COULD just change a .3dsx name to boot.3dsx, that would work ig

Sent from My iPad (Outlook for iOS<https://aka.ms/o0ukef>)

Anonymous941, 2025-04-28:
@MiksFish That wouldn't let you use a key combo...

MiksFish, 2025-04-29:
haven’t we already established that there is no possible way to use a key combo? im just giving an idea if you want something to boot when you want to use a .3dsx on boot (you would have to rename the .3dsx AND turn on autobot homebrew launcher)
anyways you can either boot it like how i said or just autobot into the homebrew launcher i don’t exactly see why you would want to autoboot into a .3dsx anyways
can someone PLEASE close this and tell this person to go to luma3ds or smth