derrekr/fastboot3DS #57: [Feature Request] Please leave boot9, boot11 and otp in memory

Adamillo, 2023-07-09:
This change would make it so that GodMod9 could not only dump those files but also allow it to use a key dumping script that dumps keys for use in Citra

Ixolaro, 2023-07-16:
i agree

profi200, 2023-07-16:
It's been a while since i discussed it with derrek but the problem is this would require us to make a number of breaking changes. We would have to permanently enable superhax which means no more chain loading fb3DS. We would need to ensure the memory regions where the dumps reside are never used and either we let FIRMs overwrite them silently or we would have to restrict allowed FIRM section ranges further.

Adamillo, 2023-07-16:
I see, that makes sense. I guess a compromise can be made in some way or another, but a solution to this problem would be good

MidyGamy, 2025-03-25:
hope this can make it