derrekr/fastboot3DS #59: Enable dev mode?

Ixolaro, 2024-04-25:
Hey, i saw that some features require "dev mode" but do i need to recompile everything with a "dev flag" or smth? if so, could someone point me out every library needed to compile on a fresh Ubuntu system? (or macos or windows)

derrekr, 2024-04-25:
To enter developer mode, navigate to the Credits section in the menu and press the Konami code button sequence.
This is currently not documented, as unexperienced users might end up bricking their system with the unlocked features.

closed 2024-04-25 by derrekr

Anonymous941, 2024-04-28:
How do you turn it off?  I don't see anything in the config file and pressing the sequence again just turns it on again

derrekr, 2024-04-28:
There should be a `"DEV_MODE"` key in your config file.